adafruit/Adafruit-Eagle-Library #16: JST-PH-2-SMT-RA positive and negative labels reversed?

jessehamner, 2015-04-27:
It appears that the "positive" and "negative" labels on the surface-mount right-angle 2-pin JST (male) jack are reversed, compared to the other three 2-pin JST jacks. I request the labels be changed.
[image]
[image]

syncon303, 2015-04-27:
No changes are necessary. If you refer to the connector datasheet (http://www.jst-mfg.com/product/pdf/eng/ePH.pdf , page1 at the bottom) you'll notice that pin 1 is on opposite side for the two versions of the connector.

jessehamner, 2015-04-28:
Though it's a bit confusing for JST to switch polarity like that, I accept that explanation. Thanks very much for replying.

closed 2015-04-28 by jessehamner