adafruit/Adafruit-Eagle-Library #19: Doesn't work

SXRguyinMA, 2015-07-25:
I downloaded the most recent files (7/24/15) and followed the instructions. When I try to open the library in Eagle 7.1.0 I get the following error:

"Loading D:/Documents/external_lbrs/adafruit.lbr ...

Error:

line 5, column 41: This is not an EAGLE file."

closed 2015-07-25 by SXRguyinMA

The-Geared-Hacker, 2016-01-13:
Did you open it like you would a project or did you open it as a library? They're two different things. Try opening it as a library.

SXRguyinMA, 2016-01-13:
I installed it as a library. I did the same with the Sparkfun library and
it worked fine

ToonSpin, 2016-12-23:
This happened to me just now, and it turns out I was not being very smart, and tried to download the library directly from Github by right-clicking. You want to either:

1. clone the Github repo and copy the file
2. download a zip and move or copy the file
3. make sure you download the "raw" version if you decide to download from Github instead.

ToonSpin, 2016-12-23:
Also @SXRguyinMA you closed this issue when you made it, but then commented half a year later implying the library still doesn't work.

If this is still an issue for you, you should not close it. Everyone including Adafruit will rightly ignore it. If it isn't, please consider adding your solution to it as a comment, so others can benefit from your experience.