adafruit/Adafruit-Eagle-Library #23: 74*125 package

buelowp, 2015-11-12:
I'm trying to use the 74*125 components for a little circuit I'm working up, but it seems the eagle library has these as individual transistors, not as the full DIL-14 package. I'm not very good with Eagle yet, so I'm not exactly sure if this is on purpose, or it's just an oversight. The 74*245 seems to be correct with pinouts and form though, so I think this is just a bug. Can someone let me know if that's true, or if it's right, how to use this form? Thanks.

uChip, 2015-11-12:
The 74*125 is a single package containing four independently controlled tri-state buffers.  To use each buffer select the 74*125 from Add... menu or button (choosing your preferred package option).  Then click on your schematic where you want to draw.  Each time you click Eagle will drop a single buffer.  If you watch the pin numbers you will see that each buffer has different pins.  You will also see that each buffer will be named the same with an "A", "B", "C" or "D" suffix calling out the individual gate in that package.  Don't forget to add the power pins by using the Invoke button or menu.
Have fun!
  - Chip

buelowp, 2015-11-12:
Thank you for the feedback, I'll do that.

I haven't checked (at work now), but will there be a valid 14 pin chip available for the .brd file? How do I get that if not?

buelowp, 2015-11-12:
Sorry, I should just add I just bought the 74AHCT125 from you guys, and want to add it to an existing circuit I have in Eagle. It would be nice if there was a defined package for it unless I'm missing something.

uChip, 2015-11-12:
The three options underneath 74\*125 are all 14 pin dip packages so if that is what you bought, you are good.  Most 7400 series parts have the same pinout across families (LS, C, AHCT, ...)(but not always across different packages: DIP, SOIC, etc), but you should always check to make sure.  In this case, I checked and they do have the same pinout.  The package will be added to the board file when you add the first gate into the schematic.
BTW, I don't work for Adafruit.  I was just being helpful.  ;-)
  - Chip

LGKev, 2017-04-01:
I am so glad that I found this. "Invoke" is what I needed to correct my board! Thank you so much

afreeland, 2019-03-01:
I am also very new to creating schematics and PCB..as in first time.  Here is a `fritzing` of how the the `74AHCT125` is wired up would love some advice as to how this maps to the transistor symbol?  The blue is connected to `GPIO 18`.
[image]

This is all I have currently setup for the Pi Zero W
[image]