adafruit/Adafruit-Eagle-Library #28: ESP8266 pinouts

edsammy, 2016-07-18:
GPIO pins 4 and 5 should be flipped  (GPIO5 --> 14, GPIO4 -->13)

xsrf, 2017-09-29:
Still an issue today :/
I don't like that libraries like this now come "integrated" with Eagle but haven't seen an update for 2 years now. Adafruit has lots of ESP8266 boards - I guess they don't use their own library.

johnboiles, 2019-10-16:
This has tripped me up twice now. In both cases I needed to revise the boards.

Looking at the schematic for their [Huzzah ESP8266](https://github.com/adafruit/Adafruit-Huzzah-ESP8266-Basic-Breakout-PCB) board, looks like they didn't correct the part they just rename the nets.

[image]

ericwertz, 2019-10-17:
> Looking at the schematic for their [Huzzah ESP8266](https://github.com/adafruit/Adafruit-Huzzah-ESP8266-Basic-Breakout-PCB) board, looks like they didn't correct the part they just rename the nets.

Theoretically they might have even updated the library, and the board is old enough to be using the the original, incorrect part.

The problem might simply be that Adafruit hasn't bothered to update the library in GitHub, and the one that we have access to here has bugs that may have been fixed internally.

I asked a few weeks ago during an Ask an Engineer if they could update this library.  They acknowledged that it was out-of-date, but didn't seem motivated to update it for whatever reason.  The response was to use the new-ish Design Block feature of Eagle to rip the part out of a schematic in which it's used, and to make it our own library component.  Sounds like they don't want to maintain the library for external consumption any more, which is their choice.  Basically, the board-level schematic is "good enough".

I've just been making all of my own library components that Adafruit also uses since this library was abandoned.  The microbuilder library is in the same state, but even worse (at least the last time I checked), so I gave up on it also.

johnboiles, 2019-10-17:
> Theoretically they might have even updated the library, and the board is old enough to be using the original, incorrect part.

I checked the library yesterday. It’s incorrect both in the latest version of the library as well as the schematic files themselves.

ericwertz, 2019-10-17:
The annotation in the file list shows that adafruit.lbr hasn't been updated in four years, so we'll never know if their own copy has bug fixes in addition to just new components or not.

As I said, their workaround was to extract the part out of board files.  I decided for myself that it made more sense to just re-create the part from scratch, but I've done enough of my own symbols/footprints by now that that goes pretty quickly.  It also gives you the chance to tweak the footprint (pads and mask) for your own process (placement tolerance, hand-solder vs reflow dimensions, stencil thickness, etc.).

The Eagle folks have always said that we can't/shouldn't trust third-party components (even the ones that they distribute), and that at the very least, they need to be reviewed before use.   Very sound advice.

Good luck!