adafruit/Adafruit-Eagle-Library #41: No more updates?

kikolobo, 2018-06-27:
This is an essential service. I really miss the newer devices

jmbldwn, 2018-06-27:
I made this a while back which you might find useful:  http://schematicpal.com

I'm about to remove the ads and make this a community-supported service.  Any suggestions for improvements are welcome.

kikolobo, 2018-06-29:
Tried checking it out but doesn’t show anything in my iPad.

I am looking for the feather footprint for a PCB I am working on, anyone has an idea where to look for it, can’t find it.

jmbldwn, 2018-06-29:
If we get enough requests, we’ll see if we can fix it for iPads.

We have “feathering” in the database.  Is that what you’re looking for?

rahmanshaber, 2020-02-02:
where to get the STEMMA_I2C_QT (STEMMA_I2C) and other new adafruit connectors ?