adafruit/Adafruit-Eagle-Library #42: WS2812B3535 footprint incompatible with datasheet

rohinkumar, 2018-09-17:
The LED3535 package from WS2812B-Mini (the so-called neopixel mini) with 3.5mm x 3.5mm footprint PCB footprint seems to be incompatible with the datasheet. The silkscreen notch line (apparently pointing pin 1) causes some confusion with LED notch which represents "GND". Also, the naming of pins is a mismatch if we put the LED on the footprint. If we rotate the pins it matches with datasheet, but the footprint is not suitable to keep the LED on top. I am not sure which one is correct- your footprint or datasheet. Please clarify the same.

mattncsu, 2019-10-29:
I think the footprint might be for the SK6812 3535 chips that they sell as mini neopixels (https://www.adafruit.com/product/2686).  The footprint is incompatible with true WS2812b-mini (http://www.normandled.com/upload/201607/WS2812B%20Mini%203535%20LED%20Datasheet.pdf) chips.
It is misleading that they put it as an alternate package under the WS2812 in the library.  Just got harsh reminder to always check footprints from online sources and 30 PCBs to practice hot air rework techniques with.

nardev, 2020-02-02:
I had same problem, the pads were in different sides.

nardev, 2020-02-11:
p.s. if you don't want to make new libs, i think that this one would fit properly: https://github.com/lionello/Lio-Eagle-Library