adafruit/Adafruit-Eagle-Library #47: Not updated since 2015?

jshep321, 2022-01-25:
Hi,
I'm looking to put some adafruit parts into my eagle design, but the library looks about 7 years old?  I'm curious about this. I'm happy to support adafruit and pay for the ease of use... but if I have to dig around to find symbols/footprints for simple parts (such as microUSB connectors), I might as well start the design sourcing parts from digikey or mouser.   Ironically sparkfun has a microUSB in it's library, but not on their website. :)

Am I missing updates somewhere? Library management for my prototypes a big pain point for me.  

Thanks!

ericwertz, 2022-01-26:
I'm pretty sure that Adafruit's take at this point is that you can extract the components from an existing schematic that uses the part -- this has been stated more than once in shows that they do (like AAE).  There's some mechanism on Eagle to do this now I believe -- although I don't know if it's an Eagle script or a built-in feature of the editor itself.  I'm just making my own.

I found this stance discouraging, but it is what it is.  I find it a bit difficult to believe that that's what they do in-house.  It may be the incremental effort for them to keep their library squeaky-clean for outside users is more effort that they want to be saddled with.

-e

jshep321, 2022-01-27:
Thanks for the detail.  My thinking was the same--they likely use internal library files and just lack the bandwidth or desire to organize and package into their library.   But! that encourages me to buy these smaller super-high-markup components from them, so might be worth it?